dronekit/dronekit-android #24: Feature Request: Connect many apps on first connect

squilter, 2015-01-11:
For dp-pebble I was hoping to wait for the user to connect using droidplanner.  Then once the user connected, dp-pebble would automatically get connected as well.

ne0fhyk, 2015-01-20:
@squilter the piggyback implementation is being added to version 1.1.8: https://github.com/DroidPlanner/3DRServices/pull/29

There's a two ways to piggy back off another app connection.
* You can register a receiver to listen to the [ACTION_VEHICLE_CONNECTION](https://github.com/DroidPlanner/3DRServices/blob/4aa4cf45effab4706335280c610a4fc806e3eb72/AidlLib/src/com/o3dr/services/android/lib/gcs/event/GCSEvent.java#L19) broadcast. That broadcast's extras contain the id of the app that connected, as well as the connection parameter used (minus the droneshare account info). Similarly, registering a receiver for [ACTION_VEHICLE_DISCONNECTION](https://github.com/DroidPlanner/3DRServices/blob/4aa4cf45effab4706335280c610a4fc806e3eb72/AidlLib/src/com/o3dr/services/android/lib/gcs/event/GCSEvent.java#L30) will let you know when an app disconnects. Examples for this mode can be found in: 
  1. [broadcast registration in manifest file](https://github.com/ne0fhyk/DP-Wear/blob/d0121cd8e95a6435a2c2f85ca082ad25cb8722fe/mobile/src/main/AndroidManifest.xml#L28-L34)
  2. [Receiver implementation](https://github.com/ne0fhyk/DP-Wear/blob/d0121cd8e95a6435a2c2f85ca082ad25cb8722fe/mobile/src/main/java/com/o3dr/android/dp/wear/receivers/GCSEventsReceiver.java)

* When your activity or service starts, and is connected to the `ServiceManager`, you can query the list of connected apps by using `ServiceManager#getConnectedApps()`. It'll return an array of bundle containing the same type of data as in the broadcast extras ([example](https://github.com/ne0fhyk/DP-Wear/blob/d0121cd8e95a6435a2c2f85ca082ad25cb8722fe/mobile/src/main/java/com/o3dr/android/dp/wear/services/DroneService.java#L139-L149)).

With that additional functionality, you might be able to remove the need for your app to run at boot, or all the time.

closed 2015-01-20 by ne0fhyk

ne0fhyk, 2015-01-20:
@squilter btw, version 1.1.8 is not out yet, so you might need to run a dev version to test the changes mentioned above.

squilter, 2015-01-24:
@ne0fhyk What's the best way to build this project?  `./gradlew installDebug` gives me
```
A problem occurred configuring project ':AidlLib'.
> SDK location not found. Define location with sdk.dir in the local.properties file or with an ANDROID_HOME environment variable.
```

I couldn't figure out how to get it into Android Studio.

ne0fhyk, 2015-01-24:
@squilter you should import the project in android studio by selected the `settings.gradle` file in the root directory.
The project will have several modules include `AidlLib`, `ClientLib` and `ServiceApp`.

`ServiceApp` is the module that needs to be compiled, and installed on the device. That module has a dependency on `AidlLib` so it'll be automatically linked in.
You should be able to do that from within android studio, without needing to run `gradlew` manually.

squilter, 2015-01-24:
Got it.  Thanks.