dronekit/dronekit-android #35: Review the 3DR Services guide

ne0fhyk, 2015-01-30:
The guide is available @ https://droidplanner.github.io/3DRServices/_guide_doc/html/index.html.

The source code for the documentation is available under the [documentation branch](https://github.com/DroidPlanner/3DRServices/tree/documentation).

ne0fhyk, 2015-01-30:
@squilter would you mind taking a look as well, and giving some feedback.

squilter, 2015-01-31:
I like it!

The "hello drone" page has a dead link at the bottom.  Points to 
https://github.com/3drobotics/Android-DroneAPI-starter

In the "getting started" page, could you elaborate a bit more about how adding `compile 'com.o3dr:3dr-services-lib:2.1.+'` to `build.gradle` works?  Does it fetch this library from the internet?

Also, what do I need to do in order to get a newer 3DR Services Client library in Android Studio?

ne0fhyk, 2015-02-02:
@phmagic I updated the content of the guide to match the latest api change. Could you update this screenshot as well to match the client lib version: https://droidplanner.github.io/3DRServices/guide_doc/html/_images/hellodrone_setup_5.png.

ne0fhyk, 2015-02-02:
@squilter The dead link points to the sample app, and should be live at the same time as the guide.

Android projects built in eclipse typically makes use of an `ant` build file behind the scenes. In Android Studio, the build tools were switched to gradle and made more explicit.
So now, for each module of an android project, you have a `build.gradle` file that defines the project configuration as well as its dependencies: https://developer.android.com/tools/studio/index.html

Any entries in the `build.gradle

Thalek, 2015-02-03:
The link in the first post appears to be broken.

ne0fhyk, 2015-02-03:
@Thalek the guide is now available @ https://droidplanner.github.io/3DRServices/.

I updated the top link to reflect the change.

Thalek, 2015-02-03:
Thank you.  I confuse easily.  :-)

closed 2015-02-09 by ne0fhyk