dronekit/dronekit-android #44: ClientLib support for gimbal attitude

alextel69, 2015-02-07:
I would like to be able to obtain gimbals roll, pitch, yaw

alextel69, 2015-02-09:
Any options?

ne0fhyk, 2015-02-09:
@alextel69 looking into it.

alextel69, 2015-02-10:
@ne0fhyk Best solution from my viewpoint would be creating Gimbal class, then obtaining via getGimbal.

ne0fhyk, 2015-02-12:
@alextel69 I'm running out of time to implement this one. Would you be able to provide a PR with an implementation similar to the `SET_RELAY` api addition?

alextel69, 2015-02-14:
@ne0fhyk good idea! Will this consider .aidl and server library?
By the way, I need not only mission, but nonmission (command) access as well

squilter, 2015-06-23:
@ne0fhyk am I modeling it after [this](https://github.com/DroidPlanner/DroneKit-Android/commit/a29ebcbdec4039b1d3bd31d27f14b862597de501)?

ne0fhyk, 2015-06-24:
@squilter yea.

hamishwillee, 2015-06-24:
@mrpollo - this would be useful for DroneKit-Python.

ne0fhyk, 2015-06-25:
Fixed by #173

closed 2015-06-25 by ne0fhyk