dronekit/dronekit-android #45: Add support for SetRelay

alextel69, 2015-02-09:


ne0fhyk, 2015-02-09:
@alextel69 I'm assuming you're referring to http://copter.ardupilot.com/wiki/mission-command-list/#do-set-relay?

alextel69, 2015-02-09:
Yes, exactly

ne0fhyk, 2015-02-09:
@alextel69 sounds good. Are you looking to use it as part of a mission?
If so a `MissionItem` implementation should suffice similar to what was done for [DO_SET_SERVO](http://copter.ardupilot.com/wiki/mission-command-list/#do-set-servo): https://github.com/DroidPlanner/3DRServices/blob/master/AidlLib/src/com/o3dr/services/android/lib/drone/mission/item/command/SetServo.java

alextel69, 2015-02-09:
I need both mission and nonmission control. I believe adding SetRelay  is a better solution then creating the class in my app

ne0fhyk, 2015-02-10:
@alextel69 `SET_RELAY` mission item support has been added in [version 1.2.5.beta](https://github.com/DroidPlanner/3DRServices/pull/49).

ne0fhyk, 2015-02-10:
Javadoc for the `SetRelay` mission item: https://droidplanner.github.io/3DRServices/javadoc/

alextel69, 2015-02-10:
@ne0fhyk Amazing!
Only nonmission control left. Possible option is to  get MavLink message from SetRelayImpl, then send it as a raw message. But it won't be a good design anyway. Waiting for command addition to drone class.

ne0fhyk, 2015-02-12:
@alextel69 the nonmission `SET_RELAY` control was added in [`version 1.2.6.beta`](https://github.com/DroidPlanner/3DRServices/pull/51).

Given that function has no defined interaction model yet, it has been placed in the `ExperimentalApi` section of the client library: https://github.com/DroidPlanner/3DRServices/blob/87ea75bdd3aab4ac7a039e229686955158121261/ClientLib/mobile/src/main/java/com/o3dr/android/client/apis/drone/ExperimentalApi.java.

To make use of the new api, you'll need version `2.2.11` of the client library: https://bintray.com/fredia/maven/3dr-services-lib/2.2.11/view

closed 2015-02-12 by ne0fhyk