dronekit/dronekit-android #82: Warning parsing

alextel69, 2015-03-14:
When MAVLINK_MSG_ID_STATUSTEXT is received, 3DRServices simply brings its message to developer.

Instead, it would be useful for developer to get PreArm, Arm etc. errors reasons by enum, thus omitting parsing step.

ne0fhyk, 2015-03-14:
This is a good point.

However, the alert broadcast mechanism is being revamped for a future
version of ardupilot (plane, copter), and so this might end up as a
redundant duplication once that work is complete.

This still could be useful to support older versions of the autopilot
without the revamped code, and to smooth out the differences between the
different versions of the autopilot (plane, copter, rover).

alextel69, 2015-03-14:
@ne0fhyk Could you explain what is going on in Ardupilot in error handling field?

Thalek, 2015-03-14:
For the record, "redundant duplication" is itself a redundant duplication.  [grin]

ne0fhyk, 2015-04-16:
@alextel69 The issue is fixed by PR #118.
 
Regarding my above comment, as mentioned in [this issue by Randy](https://github.com/DroidPlanner/Tower/issues/1329) ,arducopter will be reporting errors numbers (enums) instead of strings. When that happen, we might just mirror the errors they made available in the api where it make senses.

closed 2015-04-16 by ne0fhyk