dronekit/dronekit-android #94: Unclear disconnect flow

alextel69, 2015-03-21:
Per guide, one needs to listen to onStart/onStop events and disconnect during onStop. However, when configuration changes, onStop is called, so that drone is disconnected, while app still requires connection to drone. From user perspective it could be loss of drone control from, say, orientation change.

alextel69, 2016-07-21:
@ne0fhyk any possible solution/approach?

ne0fhyk, 2016-07-28:
@alextel69 this is more of an Android lifecycle issue. The solution in that case is to keep the drone connection state/logic in an Android component that's not subject to the Activity lifecycle. This can be a [Service](https://developer.android.com/reference/android/app/Service.html), or an [Application](https://developer.android.com/reference/android/app/Application.html) component.

closed 2016-07-28 by ne0fhyk