dronekit/dronekit-android #104: DOC: Better wording to describe "3DR Services" on http://android.dronekit.io/

hamishwillee, 2015-04-02:
http://android.dronekit.io/ or [index.rst](https://github.com/DroidPlanner/DroneKit-Android/blob/develop/doc/index.rst) 

The second paragraph should be swapped with the last paragraph (is "least important" information for developers)

The word "guide" in the second paragraph isn't quite right because it implies "instructions" rather than "listing"

    The guide to apps built with DroneKit can be found 
    on the Google Play Store under the name: 3DR Services.

Catalog (or even listing) would be better. So:

    The catalog of apps built with DroneKit can be found 
    on the Google Play Store under the name: 3DR Services.

or 

    A catalog of all DroneKit apps on the Google Play Store can be found in 3DR Services.

hamishwillee, 2015-04-02:
Fixed #105

ne0fhyk, 2015-04-02:
Fixed by #105

closed 2015-04-02 by ne0fhyk