dronekit/dronekit-android #107: Doc: Delete the built HTML in the repo?

hamishwillee, 2015-04-02:
It is not normal to also upload the built HTML in the same tree - is this intentional?

Propose changes:

* Delete the build tree from the repo
* Add the build tree to .gitignore
* Add *~ to .gitignore so temporary files are not shown in git

mrpollo, 2015-04-02:
Agreed, only that the current doc build process needs to be looked at again, @ne0fhyk can we talk offline before we decide on what to do on this ?

ne0fhyk, 2015-04-02:
@hamishwillee the built html was needed before so that they could automatically be uploaded to the gihutb pages branch. Now that everything is on the DroneKit site, they can be removed.

@mrpollo correct me if i'm wrong, but your update script only needs access to the `rst` files, right?

mrpollo, 2015-04-02:
yes, you can remove the _build folder

hamishwillee, 2015-04-03:
Setting as closed - this is addressed by #108

closed 2015-04-03 by hamishwillee