dronekit/dronekit-android #109: DOC: Setting up SITL on linux does not cover connecting using UDP in Tower

hamishwillee, 2015-04-07:
FYI some confusion from this user on getting SITL set up: http://android.dronekit.io/getting_started.html#setting-up-sitl-on-linux

From user (to Phu):

"I think if the steps you listed above were included in http://android.dronekit.io/getting_started.html#setting-up-sitl-on-linux, it would clear up most of the confusion. There are some instructions on setting up SITL in those docs, but more guidance would certainly be helpful. The missing part for me was that I didn't know I had to connect using UDP in Tower."

The instructions that were provided additionally:

"UDP is the default type for Mavlink messages to get through.

Have you tried running SITL with an explicit --out IP address?

Steps: Connect your SITL machine and your Nexus to the same Wi-FI network.

Navigate to ardupilot/ArduCopter

Run:

sim_vehicle.sh -L 3DRBerkeley --console  --out <MOBILE_DEVICE_IP>:14550

This says: run SITL with a default location of 3DRBerkeley and pip it out to MOBILE_DEVICE_IP.

Then on your Nexus, use a UDP connection, no need to specify the IP of vehicle.
"

guantau, 2016-06-06:
I've got the same problem. Is there any solution?

hamishwillee, 2016-06-06:
@guantau Yes, the solution is provided in the comment above https://github.com/dronekit/dronekit-android/issues/109#issue-66774576 -

guantau, 2016-06-07:
I have done all the steps as the comment says, but the "Hello Drone" app still can't connect to the SITL.