dronekit/dronekit-android #112: Doc: Add vagrantfile and documentation to support contribution

hamishwillee, 2015-04-09:
Copy and update the documentation support from DroneKit-Python. This includes the vagrantfile to automate docs building and a contribution page.

@ne0fhyk - I have assigned to you for the vagrantfile updates (chase @mrpollo for any help needed). When those are complete please re-assign to me and I will add/update the contributing page. OK?

ne0fhyk, 2015-04-09:
@hamishwillee I sent the instructions to @mrpollo and he's updating the vagrant file.

mrpollo, 2015-04-09:
I'll submit a PR on a branch where @hamishwillee can push his work on docs, I'll update when this is up

mrpollo, 2015-04-17:
@ne0fhyk I think I need your help on the Android SDK part https://github.com/DroidPlanner/DroneKit-Android/blob/vagrant_docs/Vagrantfile