dronekit/dronekit-android #115: dronekit client library not published

meavydev, 2015-04-12:
The documentation for DroneKit says to use the client library via:
compile 'com.o3dr:dronekit-android:2.2.+'

But the only libraries on http://jcenter.bintray.com/com/o3dr/ are 3dr-services-lib.

I have tried a local build of dronekit but that ends up with missing com.o3dr.android.client imports for ControlTower, Drone etc. although  com.o3dr.services.android.lib imports are OK.
Admittedly I am trying to extract the AAR to use it in Eclipse, but this was working for the 3DR Services Lib.

ne0fhyk, 2015-04-13:
@meavydev You can find the latest available version of the client library @ https://bintray.com/fredia/DroneKit/dronekit-android/2.2.19/view.

closed 2015-04-13 by ne0fhyk

meavydev, 2015-04-13:
OK, but that’s not the default for Android Studio...

ne0fhyk, 2015-04-13:
@meavydev we omitted adding the new library to the jcenter maven repo. The error has been corrected, and it should be available fairly soon.