dronekit/dronekit-android #117: DroneKit Client Library Eclipse Support

meavydev, 2015-04-15:
i have been trying and failing to get the Client library working in Eclipse.
Eclipse doesn't like AAR files, so I have been extracting the contents, but in the latest DroneKit this gives missing references.
It appears that the AidlLib.jar is the problem. It has grown from about 500K to 2.5MB and seems to have an AndroidManifest as well as the Android Support compatibility library.
I thought I could try building a client library in AndroidStudio, but even this is a problem as it doesn't like the AidlLib.jar due to the support lib and manifest. 
I tried removing the Android compatibility lib from my Studio test app and it still fails with the manifest:
Error:duplicate files during packaging of APK /Users/shawn/AndroidStudioProjects/dronekit/app/build/outputs/apk/app-debug-unaligned.apk
	Path in archive: AndroidManifest.xml
	Origin 1: /Users/shawn/AndroidStudioProjects/dronekit/app/build/intermediates/resources/resources-debug.ap_
	Origin 2: /Users/shawn/AndroidStudioProjects/dronekit/app/build/intermediates/exploded-aar/com.o3dr/dronekit-android/2.2.19/libs/AidlLib.jar
You can ignore those files in your build.gradle:
	android {
	  packagingOptions {
	    exclude 'AndroidManifest.xml'
	  }
	}

Now I could continue trying to get this to work, but I think the AidlLib.jar build is just not right, as it was just classes before.

ne0fhyk, 2015-04-15:
@meavydev I was able to reproduce the error. I should be able to push a fix by tomorrow. Thanks!

ne0fhyk, 2015-04-24:
@meavydev That issue took me longer than expected to diagnose :( .

Everything should be fixed by now. The client library was moved to a new repo, and can be manually retrieved from: http://jcenter.bintray.com/com/o3dr/android/dronekit-android/

The guide will be updated in the next release to reflect that change, and document the newly added apis.

closed 2015-04-24 by ne0fhyk

meavydev, 2015-04-25:
Thanks - all building again and I see the AutoPilot messages have changed, so looking forward to the doc update ;)

meavydev, 2015-04-25:
Just checked and I am getting the console messages too having worked out the changes, so thanks for that.
Of course I can't actually update my app until you release the services v1.2.13 :)

ne0fhyk, 2015-04-25:
3DR Services v1.2.13 is out in the beta channel. If no major issues is
found, it should be out in production by mid next week!