dronekit/dronekit-android #139: 3DR Services "Learn More" link is dead

squilter, 2015-05-15:
Clicking the "learn more" button links to [here](https://droidplanner.github.io/3DRServices/)

mrpollo, 2015-05-15:
@squilter where is the learn more link?

squilter, 2015-05-15:
[image]

Upper right corner.

hamishwillee, 2015-05-18:
Definitely a defect - points to non-existent doc set https://droidplanner.github.io/ . 

The problem is that it isn't clear where this should go. This is a learn-more from within the app, so I'd expect it to link to an end-user rather than developer page. We could send to the [Google play page](https://play.google.com/store/apps/details?id=org.droidplanner.services.android&hl=en) but the user will already have seen that as they have already downloaded the app. 

We could perhaps link to the [Github page for services](https://github.com/DroidPlanner/DroneKit-Android/tree/master/ServiceApp) but it is not really appropriate for end users. The [dronekit docs](http://android.dronekit.io/getting_started.html#installing-3dr-services-on-your-mobile-device) also not particularly useful for end users.

@ne0fhyk @mrpollo My response would be to point them back to the google play page for the services. If not that, Fredia, what was the intent of the link?

ne0fhyk, 2015-05-18:
@hamishwillee Since the app is mostly exposed through Tower at the moment, the intent was to lead them to the developer page as Tower's audience is (mostly) advanced.
 
That's about to shift with the upcoming solo launch however, as the bulk of the audience will be beginner and/or intermediate drone users. However I still think we can should point them to the main `DroneKit` landing page to 'educate' them about `DroneKit`, the platform powering the app they're using.

hamishwillee, 2015-05-19:
@ne0fhyk Makes sense! Fixed by #140

ne0fhyk, 2015-05-19:
Cool! It's been merged in.

closed 2015-05-19 by ne0fhyk