dronekit/dronekit-android #141: http://android.dronekit.io/getting_started.html wrong package name!

danpe, 2015-05-19:
There is a small typo but very annoying and hard to find in the tutorial:
http://android.dronekit.io/getting_started.html#setting-up-your-android-studio-project

It sais *compile 'com.o3dr.android:droneKit-android:2.3.+'*
with capital *K* on *dronKit*, it shuld be lower case *k* so it will be *dronekit-android:2.3.+'* otherwise *grade 1.2.3* wont find it on jcenter() repository.

Took me a while to figure.

mrpollo, 2015-05-19:
thanks for the report @danpe thats a nasty one, I'm sure @hamishwillee can get this sorted out

hamishwillee, 2015-05-19:
Fixed by #142. I haven't actually tested that @danpe change "works" - just changed the text. It seems very logical that it would work though :-)

mrpollo, 2015-05-20:
thanks a lot @hamishwillee I think @ne0fhyk will test before merging

hamishwillee, 2015-05-20:
@mrpollo @ne0fhyk - Just to be very precise, I did test the docs build. What I haven't tested is building an Android project with the new text.

ne0fhyk, 2015-05-20:
Tested and merged !

PR #142 (2015-05-20): Fixes incorrect dronekit package name
Fixes  #141
closed 2015-05-20 by ne0fhyk