dronekit/dronekit-android #151: Arming while RC_Failsafe should not allow the arm function to proceed.

Djacob502, 2015-06-01:
To produce this issue first turn the RC_Transmitter off: 

When the arm command is first sent and the ErrorType - RC_Failsafe is returned correctly and the drone does not arm. 

If the arm command is then sent a second time - the ErrorType - RC Failsafe is not returned and the drone does arm even through the RC_Transmitter is still off.

squilter, 2015-06-23:
@ne0fhyk I can easily put an if-statement [here](https://github.com/DroidPlanner/DroneKit-Android/blob/a1bc2c5baedf5fe51b7f58448e0c964dbc102cf1/ClientLib/src/main/java/com/o3dr/android/client/Drone.java#L484).  But I'm not sure what to check for.  I can't reject the arming request if there is no controller - some people fly with only Tower (no controller).  I also can't check to see if we are in a failsafe state - failsafe isn't a state.

Maybe this would be better for flight code?

squilter, 2015-07-07:
I was able to reproduce the issue.  Since anything related to arming is a safety issue, the best place to fix this is in flight code.  I have made a new issue at diydrones/ardupilot#2529

closed 2015-07-07 by squilter