dronekit/dronekit-android #153: 【Compile Error】Can you tell me how to solve it?

Linjieqiang, 2015-06-03:
Hello,Thanks for your work.But I want to express,why I always failed when I try to compile this project.And the error information is that " Error:Execution failed for task ':ServiceApp:preDexDebug'.
> com.android.ide.common.process.ProcessException: org.gradle.process.internal.ExecException: Process 'command 'C:\Program Files\Java\jdk1.8.0_45\bin\java.exe'' finished with non-zero exit value 1"
Can you tell me how to solve it?I am looking forward to your answer.

ne0fhyk, 2015-06-03:
Could you attempt to compile using java 7 instead.

Linjieqiang, 2015-06-05:
But I compiled Tower is fine . Did you compile using  java 7 ?

Linjieqiang, 2015-06-12:
OK ! I am excited to feedback that when I compile with java7 ,it's no error ! Thanks !

closed 2015-06-12 by Linjieqiang