dronekit/dronekit-android #159: Setting autopilot data stream rates

maximbatalin, 2015-06-05:
Currently 3DR services seems to be resetting the autopilot stream rates to 2Hz. As I've heard from the support team there is no way of changing this now. I would like to kindly ask you to please consider adding a functionality that would allow setting a desired data stream rate.

In my case this was the issue when connecting a sensor to the pixhawk and needing a data rate of at least 10Hz. Upon changing data rate via APM Planner, the data rate would revert back to 2Hz when accessed through dronekit. Therefore, this fix, I believe, is very important for all of the practical applications that need to be able to process data at high data rates.

Thank you in advance for consideration and fixes.

ne0fhyk, 2015-06-06:
@maximbatalin until this issue is resolved, you can set the default stream rate by updating the [`DroidPlannerPrefs#getRates() method`](https://github.com/DroidPlanner/DroneKit-Android/blob/dd5bc6d8f3fc72e9c174ad897dc8a71bad018716/ServiceApp/src/org/droidplanner/services/android/utils/prefs/DroidPlannerPrefs.java#L68-L86).

Let me know if you need a complete set of instructions for building the app after the update is performed.

maximbatalin, 2015-06-06:
Thank you very much for a follow up. If you could leave a full set of instructions - that would be most appreciated.

Also - this method is getting the rates. Is it then used internally by 3DR service to throttle the parameter read period?

ne0fhyk, 2016-07-28:
This is now possible through the [`VehicleApi#updateVehicleDataStreamRate(...)`](https://github.com/dronekit/dronekit-android/blob/b3101dbe7ff42c2408248ab2cc820ff8ad7420ce/ClientLib/src/main/java/com/o3dr/android/client/apis/VehicleApi.java#L261) method.

closed 2016-07-28 by ne0fhyk