dronekit/dronekit-android #163: For Hello Drone app, getting Drone Disconnected/ Connection Failed: only on emulator

csathe, 2015-06-06:
Hello, 

I am getting Drone Disconnected/ Connection Failed: only on emulator  for the Hello Drone app.  I confirmed details like correct ip while launching STIL simulation, STIL simulation correctly running (Ready to FLY, received 510 parameters) etc. The same setup works on real Android devices. 
For emulator I tried using both 127.0.0.1 and the ip listed in About Device on the Android emulator. (in below screenshot 10.0.2.15 is the ip listed in About Phone on the emulator)
Am I missing something ? Can someone please help with a potential solution ? 

[image]

[image]

[image]

ne0fhyk, 2015-06-08:
I have yet to successfully be able to get the default android emulator taking to sitl on the local network. 
Instead, I have used genymotion when testing via emulation.

squilter, 2015-06-26:
I don't think this can be done through the emulator without any complicated port forwarding.  I just booted up the emulator, and it didn't have an external ip-address on the LAN.  It forwarded in some weird way through the computer, so I couldn't get packets there.

As @ne0fhyk said, genymotion does work.  It just opens op android in a virtualbox, which does give it a proper external ip-address.

ne0fhyk, 2015-06-29:
Closing issue as alternative solution was suggested.

closed 2015-06-29 by ne0fhyk