dronekit/dronekit-android #170: Check 3DRServices version/presence

alextel69, 2015-06-14:
Problem: 
1) No way to check presence/version of 3DRServices on device.
2) Automatic error dialog show-up on connection. 

Solution: same pattern as in Google Play Services http://goo.gl/hTFrbf

ne0fhyk, 2015-06-14:
@alextel69 there's already a mechanism to check if 3DR Services is installed. For example, try starting Tower with 3DR Services uninstalled, and you'll get a prompt to install/upgrade to the latest version.

ne0fhyk, 2015-06-14:
@alextel69 the check is done automatically, transparently to the application when the application connects to the `ControlTower` instance: https://github.com/DroidPlanner/DroneKit-Android/blob/a09a4ca2edc6ed9564bd611f3c68e7fcc41a3815/ClientLib/src/main/java/com/o3dr/android/client/ControlTower.java#L135-L152

alextel69, 2015-06-15:
@ne0fhyk automatic check/dialog is fine, however ability to conduct manual check is better, because lets you help user make decision in your way.
For instance, one may need to show message in notification bar the way Google Apps do this when Play Services need update. Especially useful for Tower, as one may install Tower, but not 3DRServices, then go to field and be unable to fly because of missing library - which is a bad UX from first minutes.
Other option is to show custom localized dialog.

ne0fhyk, 2015-06-15:
Good point. I'm working on adding a capability api to allow client apps the
ability to query which functionality they have access to. The functionality
you're mentioning could be added there as well.

alextel69, 2015-06-23:
@ne0fhyk you may take a look at how Google deals with library errors with [GoogleApiAvailability] (https://developers.google.com/android/reference/com/google/android/gms/common/GoogleApiAvailability)

ne0fhyk, 2015-07-08:
Fixed by pr #188: The [`ApiAvailability`](https://github.com/DroidPlanner/DroneKit-Android/blob/06bdbf792088fa0c8199c660324ac81dc41958de/ClientLib/src/main/java/com/o3dr/android/client/apis/ApiAvailability.java) class provides the requested functionality.

closed 2015-07-08 by ne0fhyk