dronekit/dronekit-android #203: Rename 3DR Services to DroneKit Services

ne0fhyk, 2015-07-17:


ne0fhyk, 2015-07-17:
@phmagic @hamishwillee @mrpollo @tcr3dr thoughts?

tcr3dr, 2015-07-17:
I support this, as it's less confusing for a developer. My only reservation would be if an end-user has issue with installing DroneKit Services vs. 3DR Services; would there be a huge impact there?

hamishwillee, 2015-07-17:
Yes, good idea. @tcr3dr I don't think a new user would find this an issue. More concerned about someone updating - would they need to install a new package or would this be a silent update to the "old" 3DRServices?

ne0fhyk, 2015-07-17:
Hamish, this would be an update to the old 3DR Services app.
I don't think it's much of a problem since most users don't interact with
the 3DR Services, only with its clients apps.

hamishwillee, 2015-07-18:
In that case the only issue is making sure that the transition is managed in the docs. With the actual names of file locations (e.g. to examples, github repos, etc) and package names (e.g. code libraries) change or is this just a "UI-level" change?

ne0fhyk, 2015-07-30:
From the last DK meeting, a name change wasn't judged necessary.

closed 2015-07-30 by ne0fhyk