dronekit/dronekit-android #233: Incosistency with Maven

alextel69, 2015-08-09:
Drone.changeVehicleMode is deprecated. Instead, VehicleApi should be used. 
But there is no VehicleApi class in library. 
I am using Android-DroneKit version 2.3.+

ne0fhyk, 2015-08-09:
Are you seeing the deprecation warnings in the 2.3.+ library?
The VehicleApi is within the upcoming library update (v2.5.+), but you
should not be seeing deprecation warnings within the currently available
client library.

alextel69, 2015-08-10:
I switched to 2.3.34 (latest version on jcenter) and still face deprecations alongside with miss of reccomended classes

ne0fhyk, 2015-08-14:
@alextel69 the issue should be fixed now since `dronekit-android v2.5.+` is now out and available.

closed 2015-08-14 by ne0fhyk