dronekit/dronekit-android #295: Enable SIMPLE flight mode

dgmccut, 2015-10-07:
The SIMPLE flight mode should be able to be entered by calling:

VehicleApi.getApi(drone).setVehicleMode(VehicleMode.COPTER_SIMPLE, this);

Here's the reference to SIMPLE mode
http://copter.ardupilot.com/wiki/simpleandsuper-simple-modes/#simple_mode

squilter, 2015-10-07:
Simple mode isn't actually a `mode` like stabilize or loiter.  It is a flag that can be on or off for the real `mode`s.  So Loiter/stabilize/althold/etc can be flown with normal, simple, or super-simple.  So I don't think it is appropriate to call it with `setVehicleMode();`.

dgmccut, 2015-10-07:
I agree. I'm new to all this.

ne0fhyk, 2016-01-12:
Enabling `simple` flight mode is not yet supported via mavlink. Closing this for now.

closed 2016-01-12 by ne0fhyk