dronekit/dronekit-android #302: Support bebop flight modes

ne0fhyk, 2015-10-12:
I try to perform a Land and I receive
 MAV_CMD_SET MODE base 1 target sys 1 custom 9
The bebop is expecting a MAV_CMD_NAV_LAND.

I try to perform Return Home
 MAV_CMD_SET MODE base 1 target sys 1 custom 6
The bebop is expecting a  MAV_CMD_NAV_RETURN_TO_LAUNCH.

Moreover for the manual pause command, we use MAV_GOTO_DO_HOLD command.
Currently it receives a waypoint.

ne0fhyk, 2015-10-12:
@jlbilou93 What other flight modes are supported by the bebop?

jlbilou93, 2015-10-13:
For the moment, the bebop can receive manual_control, there is no need to set a mode for this.
A flightplan mode is also possible without setting a mode. Just by sending waypoints and starting it with MAV_CMAV_CMD_MISSION_START .
The RTL is also supported by sending MD_NAV_RETURN_TO_LAUNCH