dronekit/dronekit-android #313: Battery Status is not being displayed

Thoin, 2015-10-19:
*Phone*: Samsung Galaxy S5
*3DR Services*: 1.4.3 beta 2
*Tower*: 3.2.0

@ne0fhyk, I can now successfully connect to my Pixhawk running PX4 firmware with the beta of 3DR services. It's great!

I get telemetry data in real time as expected. I do **NOT** get battery status though. Voltage and Current read 0.00, and R reads 0% and D has no data either. This is data that I get when using QGroundControl so I know my hardware is not at fault.

[3DR Service Logs](https://drive.google.com/folderview?id=0B-4Pt7stMkc8UzhzOE5uMGFqaGc&usp=sharing)

ne0fhyk, 2015-10-20:
Fixed by pr #314

ne0fhyk, 2015-10-20:
@Thoin I'll provide you with a build for testing.

closed 2015-10-20 by ne0fhyk