dronekit/dronekit-android #351: Solo - MavlinkMessageWrapper createFromParcel NoSuchMethodExceptions

meavydev, 2015-11-22:
I am not sure exactly what is causing the exception, as it's hidden, but running an Android Studio Allocation Tracking to see where my app was causing more GCs shows up  a few thousand instances of MavlinkMessageWrapper createFromParcel NoSuchMethodException.
Obviously avoiding the exception will help performance and GCs.
This is with com.o3dr.android:dronekit-android:2.6.9 connected to a 3DR Solo.

meavydev, 2015-11-27:
I have got the tracing out of one of the exceptions, but I think there might be more:
                                                           java.lang.RuntimeException: Parcelable encountered ClassNotFoundException reading a Serializable object (name = com.MAVLink.ardupilotmega.msg_gps_accuracy)
                                                               at android.os.Parcel.readSerializable(Parcel.java:2491)
                                                               at android.os.Parcel.readSerializable(Parcel.java:2456)
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper.<init>(MavlinkMessageWrapper.java:38)
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper.<init>(MavlinkMessageWrapper.java:11)
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper$1.createFromParcel(MavlinkMessageWrapper.java:43)
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper$1.createFromParcel(MavlinkMessageWrapper.java:41)
                                                               at com.o3dr.services.android.lib.model.IMavlinkObserver$Stub.onTransact(IMavlinkObserver.java:53)
                                                               at android.os.Binder.execTransact(Binder.java:453)
                                                            Caused by: java.lang.ClassNotFoundException: com.MAVLink.ardupilotmega.msg_gps_accuracy
                                                               at java.lang.Class.classForName(Native Method)
                                                               at java.lang.Class.forName(Class.java:324)
                                                               at java.io.ObjectInputStream.resolveClass(ObjectInputStream.java:2258)
                                                               at android.os.Parcel$2.resolveClass(Parcel.java:2482)
                                                               at java.io.ObjectInputStream.readNewClassDesc(ObjectInputStream.java:1641)
                                                               at java.io.ObjectInputStream.readClassDesc(ObjectInputStream.java:657)
                                                               at java.io.ObjectInputStream.readNewObject(ObjectInputStream.java:1782)
                                                               at java.io.ObjectInputStream.readNonPrimitiveContent(ObjectInputStream.java:761)
                                                               at java.io.ObjectInputStream.readObject(ObjectInputStream.java:1983)
                                                               at java.io.ObjectInputStream.readObject(ObjectInputStream.java:1940)
                                                               at android.os.Parcel.readSerializable(Parcel.java:2485)
                                                               at android.os.Parcel.readSerializable(Parcel.java:2456) 
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper.<init>(MavlinkMessageWrapper.java:38) 
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper.<init>(MavlinkMessageWrapper.java:11) 
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper$1.createFromParcel(MavlinkMessageWrapper.java:43) 
                                                               at com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper$1.createFromParcel(MavlinkMessageWrapper.java:41) 
                                                               at com.o3dr.services.android.lib.model.IMavlinkObserver$Stub.onTransact(IMavlinkObserver.java:53) 
                                                               at android.os.Binder.execTransact(Binder.java:453)

meavydev, 2015-11-29:
Mmm well it could be down to Dexguard, but I am not sure why, as it's a debug build.
Possibly it was trimming out all the mavlink classes?

However there is still one thing, which I can raise as a separate issue.
Using createFromParcel() is very memory hostile, as it creates and disposes loads of objects, so if you can avoid creating so many, it would help performance.
Have you looked at a cache ?

meavydev, 2015-11-29:
Nope - error is back so I guess it's just it didn't have a GPS fix earlier.

chaviw, 2016-04-14:
Have you tried using the latest dronekit-android client. That particular mavlink class was added in November so the build you have may not have that class

closed 2016-07-28 by ne0fhyk