dronekit/dronekit-android #369: RC Channel Override Alternative

skyritsis, 2016-01-06:
Hello. I am developing an app in which I want to move the quadcopter, while using external sensor data. This happens exclusively in landing mode. The approach I am currently following, is to use rc channel override on pitch and roll channels, but this is discouraged in the documentation. So, I would like to ask you, what is your suggestion on the specific problem.

Thank you in advance,
-SK

squilter, 2016-01-06:
Are you trying to land on something?

skyritsis, 2016-01-06:
Exactly, I am trying to land on a specific platform.
With the means I am currently using, I am able to tell if I need to alter the pitch and roll of the copter.
But I am trying to find the best way to alter them.

squilter, 2016-01-06:
You should send a `TARGET_LAND` message.  I did this a few days ago in python, so you should just need to port it over.  [Here's](https://github.com/squilter/target-land/blob/master/target_land.py) my code.

Here's a video of it working:
https://youtu.be/PURlcAu6MEk

And here's a definition of the MAVLink message:
http://mavlink.org/messages/common#LANDING_TARGET

skyritsis, 2016-01-06:
Thank you very much for your time. I will have a look at it and report my results here.
I really appreciate your help.

closed 2016-01-07 by skyritsis