dronekit/dronekit-android #370: Is message_factory included in the dronekit-android?

skyritsis, 2016-01-07:
As I was guided in a previous question, I am trying to use the message_factory to use landing_target_encode. But after some research on the documentation, I can't figure out how to do it. Am I missing something? Is this call not included in the android api?

Thank you for your time in advance,
-SK

ne0fhyk, 2016-01-08:
@skyritsis The android api is slightly different. You can accomplish the same task using the [`ExperimentalApi#sendMavlinkMessage(...)`](http://android.dronekit.io/javadoc/com/o3dr/android/client/apis/ExperimentalApi.html#sendMavlinkMessage(com.o3dr.services.android.lib.mavlink.MavlinkMessageWrapper)) method, with the mavlink message itself wrapped into a [`MavLinkMessageWrapper`](http://android.dronekit.io/javadoc/com/o3dr/services/android/lib/mavlink/MavlinkMessageWrapper.html) container.

ne0fhyk, 2016-01-08:
@skyritsis btw, the best place for these questions is the [dronekit-android gitter channel](https://gitter.im/dronekit/dronekit-android).
thanks!

closed 2016-01-08 by ne0fhyk