dronekit/dronekit-android #380: ManualControl in Control API

RazaRana, 2016-03-09:
manual control doesn't work in control api. when i call the method to change velocity nothing happen

r1ckr, 2016-04-12:
Happening to me as well with versions 2.6.9 2.9.0.

chaviw, 2016-04-14:
I know this conversation was started on gitter, but I have a few follow up questions. Were you running against a sitl vehicle or real vehicle? The callback may not come back but you should still see the vehicle moving

r1ckr, 2016-04-15:
Hey @chaviw, thanks for the reply, I am running it against a sitl vehicle and I can't see it moving in the map nor in FlightGear. Is it working for you? Maybe is a difference in the setup, mine is:

APM:Copter V3.4-dev
drone-kit versions: 2.6.9, 2.9.0
OS: Linux

RazaRana, 2016-04-15:
not working for both not in SITL nor in vehicle.

ne0fhyk, 2016-07-28:
Closing this issue for lack of activity. @RazaRana If you're still experiencing the issue, can you ping us on the gitter channel: https://gitter.im/dronekit/dronekit-android

closed 2016-07-28 by ne0fhyk