dronekit/dronekit-android #398: 3DR Services app crashes when using newTcpConnection()

arashn, 2016-04-12:
When I use newTcpConneciton() to create a TCP connection to the drone, the app connects to the drone, but then the 3DR Services app crashes immediately.

chaviw, 2016-04-13:
Can you provide the stack trace of the crash?

arashn, 2016-04-14:
This is the stack trace of the crash:

```
E/AndroidRuntime(15900): FATAL EXCEPTION: MavLinkConnection-Manager Thread
E/AndroidRuntime(15900): java.lang.IllegalArgumentException: Value is outside of the range of an unsigned byte: -1
E/AndroidRuntime(15900): 	at com.MAVLink.Messages.MAVLinkPayload.putUnsignedByte(MAVLinkPayload.java:140)
E/AndroidRuntime(15900): 	at com.MAVLink.common.msg_mission_item.pack(msg_mission_item.java:124)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.communication.service.MAVLinkClient.sendMavMessage(MAVLinkClient.java:241)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.communication.service.MAVLinkClient.sendMessage(MAVLinkClient.java:233)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.communication.service.MAVLinkClient.sendMessage(MAVLinkClient.java:42)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.WaypointManager.processWaypointToSend(WaypointManager.java:289)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.WaypointManager.processMessage(WaypointManager.java:207)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.drone.autopilot.apm.ArduPilot.onMavLinkMessageReceived(ArduPilot.java:397)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.drone.manager.MavLinkDroneManager.notifyReceivedData(MavLinkDroneManager.java:219)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.drone.manager.MavLinkDroneManager.notifyReceivedData(MavLinkDroneManager.java:62)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.communication.service.MAVLinkClient$1.onReceivePacket(MAVLinkClient.java:58)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.connection.MavLinkConnection.reportReceivedPacket(MavLinkConnection.java:499)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.connection.MavLinkConnection.access$600(MavLinkConnection.java:27)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.connection.MavLinkConnection$2.handleData(MavLinkConnection.java:169)
E/AndroidRuntime(15900): 	at org.droidplanner.services.android.core.MAVLink.connection.MavLinkConnection$2.run(MavLinkConnection.java:138)
E/AndroidRuntime(15900): 	at java.lang.Thread.run(Thread.java:856)
```

chaviw, 2016-04-14:
Are you doing anything before connecting that's related to missions? It seems like it's executing code where it thinks you've requested info about the mission on the vehicle

arashn, 2016-04-16:
No, not before connecting. I create a Mission and send it to the drone, but only after receiving AttributeEvent.STATE_CONNECTED.

chaviw, 2016-04-16:
`AttributeEvent.STATE_CONNECTED` is called when the TCP connection is established, but not necessarily if the vehicle is connected. If you want to make sure the vehicle is connected, listen for AttributeEvent.HEARTBEAT_FIRST`

arashn, 2016-04-16:
I tried that. Unfortunately, I still get the same error as above.

arashn, 2016-05-26:
@chaviw I can now confirm that this issue was due to the SITL that I was connecting to. Installing the latest version of the SITL fixed this issue. Thanks.

closed 2016-05-26 by arashn