dronekit/dronekit-android #403: Getting specific mavlink message

santiandres33, 2016-05-06:
Hey Guys,	

I am working with dronekit Android, and I am trying to get an specific mavlink message (25), although I don't why I am not getting it. Also try getting any other message but same result. Any thought maybe be very appreciated, 

msg_message_interval msgMessageInterval = new msg_message_interval();
                msgMessageInterval.message_id = 25;
                msgMessageInterval.interval_us = 10;

                MavlinkMessageWrapper mavlinkMessageWrapper = new MavlinkMessageWrapper(msgMessageInterval);
                
                Log.d("REQUESTED MAV",mavlinkMessageWrapper.getMavLinkMessage().toString());
                
                ExperimentalApi.getApi(MainActivity.drone).sendMavlinkMessage(mavlinkMessageWrapper);
                
                MainActivity.drone.addMavlinkObserver(new MavlinkObserver() {
                    @Override
                    public void onMavlinkMessageReceived(MavlinkMessageWrapper mavlinkMessageWrapper) {
                        Log.d("Received Mavlinks:",mavlinkMessageWrapper.getMavLinkMessage().toString());
//                        Toast.makeText(getBaseContext(),mavlinkMessageWrapper.getMavLinkMessage().toString(),Toast.LENGTH_LONG).show();

chaviw, 2016-07-20:
Try switching the order when you add the observer and when you send the mavlink message. 
It's always possible that the app is getting a response before you register the observer.

Also are you sure there's a response for that message?

Your code should be
```
MainActivity.drone.addMavlinkObserver(new MavlinkObserver() {
     @Override
     public void onMavlinkMessageReceived(MavlinkMessageWrapper mavlinkMessageWrapper) {           
         Log.d("Received Mavlinks:",mavlinkMessageWrapper.getMavLinkMessage().toString());
         //Toast.makeText(getBaseContext(),mavlinkMessageWrapper.getMavLinkMessage().toString(),Toast.LENGTH_LONG).show();

ExperimentalApi.getApi(MainActivity.drone).sendMavlinkMessage(mavlinkMessageWrapper);
```

ne0fhyk, 2016-07-20:
@santiandres33 In addition, make sure your `drone` object is connected and receiving telemetry before adding the mavlink observer object.
Also, I'm closing this issue, since this type of questions should be addressed to the gitter channel: https://gitter.im/dronekit/dronekit-android

closed 2016-07-20 by ne0fhyk