dronekit/dronekit-android #404: Parameters keep getting received

meavydev, 2016-05-16:
Not sure when this started happening, as I hadn't changed my app, but I just got a report that my parameters received dialogue keeps updating and doesn't stop.
This is connected to a 3DR Solo with the latest firmware and I get the first 50 parameters continuously coming through in a cycle:
05-16 09:02:59.356 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 0
05-16 09:02:59.356 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 1
05-16 09:02:59.356 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 2
05-16 09:02:59.356 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 3
05-16 09:02:59.356 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 4
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 5
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 6
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 7
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 8
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 9
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 10
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 11
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 12
05-16 09:02:59.357 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 13
05-16 09:02:59.358 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 14
05-16 09:02:59.358 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 15
05-16 09:02:59.360 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 16
05-16 09:02:59.360 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 17
05-16 09:02:59.361 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 18
05-16 09:02:59.363 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 19
05-16 09:02:59.397 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 20
05-16 09:02:59.398 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 21
05-16 09:02:59.398 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 22
05-16 09:02:59.398 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 25
05-16 09:02:59.399 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 26
05-16 09:02:59.400 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 27
05-16 09:02:59.405 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 28
05-16 09:02:59.405 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 29
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 30
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 31
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 32
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 33
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 38
05-16 09:02:59.406 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 39
05-16 09:02:59.510 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 40
05-16 09:02:59.510 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 45
05-16 09:02:59.510 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 47
05-16 09:02:59.510 24891-24891/meavydev.DronePro D/ARDroneMap: Parameter received index: 49

Have the parameter update events been changed recently?

ne0fhyk, 2016-05-16:
@meavydev Can you get the firmware version for the 3dr Solo that's experiencing this issue.
I have run into the issue as well, and from troubleshooting, found out the issue is originating from Solo, instead of the app.
I informed the firmware team of the issue.

meavydev, 2016-05-16:
According to the Solo App I have just updated to 2.3.0-1.
It seemed OK on the previous version, but I can't remember what that was unfortunately.

I guess for the moment I will just have to hide the dialogue, but the continuous parameter updates aren't going to be great.

ne0fhyk, 2016-05-16:
That firmware version is the cause of the issue. It introduces a bug that causes the autopilot to continuously emit mavlink params messages. Unfortunately, there's nothing that the client library or 3DR Services can do to resolve the issue. The firmware will have to be fixed, and updated for it to go away.

Do your app makes use of the parameter values?

meavydev, 2016-05-16:
OK thanks.
It does use the parameter values, but I will just have to hide the dialogue for the moment - it was mainly for things like joystick calibration on Iris.

ne0fhyk, 2016-07-04:
@TShapinsky can you check if this issue has been fixed in the latest version of the Solo firmware.

ne0fhyk, 2016-07-28:
@TShapinsky is this still an issue?

ne0fhyk, 2016-08-29:
Fixed in the latest version of Tower.

closed 2016-08-29 by ne0fhyk