dronekit/dronekit-android #408: Make FollowApi accept locations supplied by the client

kellyschrock, 2016-06-19:
FollowApi is limited in the ways it can be used, since it insists on only using the location of the device running the client application. It should be able to also accept an externally-supplied location from the application, so there are more options on what can be followed.

ne0fhyk, 2016-06-20:
@kellyschrock sounds like a good idea. 
What do you think about the following changes to the `FollowApi` to provide that functionality:
* Add an overloaded method for `FollowApi#enableFollowMe(...)` with an additional (to be defined) `enum` parameter to specify the origin of the location updates
* Add `FollowApi#updateFollowLocation(android.location.Location location, ...)` to provide location updates.
* Update the `FollowApi#updateFollowParams(...)` method to allow switching between different location provider (i.e: device, custom).

kellyschrock, 2016-06-20:
@ne0fhyk That sounds good. I have a branch _almost_ ready for PR that does a lot of this, but I'm currently using a boolean to indicate whether follow should be "external" or not. I like the enum idea more, actually. I can push it if you like and you can take a look.

ne0fhyk, 2016-06-21:
@kellyschrock sounds good. looking forward to the PR!

kellyschrock, 2016-06-22:
Here you go! :-) https://github.com/dronekit/dronekit-android/pull/410

Thanks,
Kelly

PR #410 (2016-07-01): Followapi update location
Allow FollowApi to use locations supplied by the app.

Fixes #408.
commit ecc10f24 (2016-07-01): FollowApi -- Allow external locations

Allow FollowApi to use locations supplied by the app.

Fixes #408.
closed 2016-07-01 by ne0fhyk