dronekit/dronekit-android #411: Verify network binding setup

ne0fhyk, 2016-06-24:
Ensure that the sockets used in the library are  bound to the right network.

ne0fhyk, 2016-07-27:
@chaviw any updates on the status for this issue?

chaviw, 2016-07-27:
I started working on it. I realized it might a bit more complex in this case because I need to make sure it's the right network it's binding to. Basically, any sockets that are opened need to know what network. Either the caller tells them or some static method in wifi manager can handle that.

Let me take a look at this week and see how much effort it might be

ne0fhyk, 2016-08-23:
@chaviw, brainstorming this, the network binding occurs before any connection is established. 
As such by the time network connections start being made, the network should already be available (if needed). So we can either pass it through the connections' constructors or define another `openConnection(Network network)` abstract method to complement the current `openConnection()` method.

ne0fhyk, 2016-08-29:
Fixed by pr #433

closed 2016-08-29 by ne0fhyk