dronekit/dronekit-android #413: Mavlink error on Build

Gamma-Software, 2016-06-30:
Hi guys ;)
I'm having a build issue when I want to build the dronekit library. The Gradle Build generate an error telling that the package of MAVLink doesn't exist. Here is the log:

[image]

Everything is related to the MAVLink Lib.

Do you guys know what's going on ?
Thanks for the help ;)

Gamma-Software, 2016-07-04:
No suggestion ?

Gamma-Software, 2016-07-04:
Well, I finally solved the issue. We just have to use a different version name :/

closed 2016-07-04 by Gamma-Software

NbagusW, 2016-07-22:
Hi @Gamma-Software , how can u solve it? i had the sam probmels  wih u, but i still didnt know howto fix it.. ty

Gamma-Software, 2016-07-25:
Make sure that the version of the Library is not conflicting with the current version of the release library (in the ClientLib Build.gradle). 
So change the version:
[image]

NbagusW, 2016-07-28:
@Gamma-Software thanks but, i still have an error :( ,where should i change the version, or to know if its already have same version. 
PS: really sorry , im a newbie at this.

Gamma-Software, 2016-07-28:
I think that the first version is 
VERSION_MAJOR = 2
VERSION_MINOR = 9
VERSION_PATCH = 0
VERSION_SUFFIX = "release"

You just have to change the version number (for instance: 
VERSION_PATCH = 1).

Otherwise, give me your message error

ne0fhyk, 2016-07-28:
@NbagusW can you post your question on the [dronekit-android gitter channel](https://gitter.im/dronekit/dronekit-android) instead.

Thanks.