dronekit/dronekit-android #419: LocationRelay needs update

kellyschrock, 2016-07-26:
LocationRelay has two issues when used with Follow in CLIENT_SPECIFIED mode:

It rejects incoming locations that don't have a heading, accuracy, and time. It doesn't actually need heading for anything, which means that applications supplying external locations to the API have to specify an arbitrary heading just to get LocationRelay to accept it. This requirement should be moved.

The currentSpeed calculation falls apart for incoming location intervals of < 1 second, returning 0 for something as straightforward as (0.5 meter distance in 500ms). This failure affects the average speed, which makes subsequent locations more likely to be rejected.

ne0fhyk, 2016-07-26:
@kellyschrock The location heading is being used when [generating the gcs location](https://github.com/dronekit/dronekit-android/blob/9fa9d695a3bebeba85467c2e366d59ce6c1ff84d/ServiceApp/src/org/droidplanner/services/android/core/gcs/follow/LocationRelay.java#L73-L83), and is then used by some of the Follow algorithm like [`FollowSplineAbove`](https://github.com/dronekit/dronekit-android/blob/9fa9d695a3bebeba85467c2e366d59ce6c1ff84d/ServiceApp/src/org/droidplanner/services/android/core/gcs/follow/FollowSplineAbove.java#L24), so it's actually needed.

ne0fhyk, 2016-07-26:
@kellyschrock as an alternative, the location relay class can detect when the received location has no bearing, and compute a bearing using [`GeoTools#getHeadingFromCoordinates(...)`](https://github.com/dronekit/dronekit-android/blob/9fa9d695a3bebeba85467c2e366d59ce6c1ff84d/ServiceApp/src/org/droidplanner/services/android/core/helpers/geoTools/GeoTools.java#L150) passing the new location, and the last received location.
It will then use that bearing when creating the gcs location if the received location has no bearing.

PR #420 (2016-07-27): LocationRelay improvement
Fixes #419.
commit 5888efce (2016-07-27): LocationRelay improvement

Fixes #419.
closed 2016-07-27 by ne0fhyk