dronekit/dronekit-android #422: 1.6.0-beta.2 update throwing BadParcelableException

DarylWM, 2016-08-07:
The 1.6.0-beta.2 update seems to be having trouble with messages. On arming my Solo, Tower crashes. I found this exception was in the crash logs (Android | data | org.droidplanner.android | files | crash_log):

android.os.BadParcelableException: ClassNotFoundException when unmarshalling: com.o3dr.services.android.lib.drone.companion.solo.tlv.SoloReturnHomeLocationMessage
	at android.os.Parcel.readParcelableCreator(Parcel.java:2432)
	at android.os.Parcel.readParcelable(Parcel.java:2358)
	at android.os.Parcel.readValue(Parcel.java:2264)
	at android.os.Parcel.readArrayMapInternal(Parcel.java:2614)
	at android.os.BaseBundle.unparcel(BaseBundle.java:221)
	at android.os.Bundle.putAll(Bundle.java:182)
	at android.content.Intent.putExtras(Intent.java:7824)
	at org.droidplanner.android.DroidPlannerApp$3.run(DroidPlannerApp.java:131)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:148)
	at android.app.ActivityThread.main(ActivityThread.java:7325)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

My configuration: Android 6.0.1, Tower 3.2.1 and 3DR Services 1.6.0-beta.2. On the Solo and controller I'm using 2.4.0.

ne0fhyk, 2016-08-07:
@DarylWM The 1.6.0-beta.2 update was not released by us, and as such we have no insight into the contents of the update.
Please contact 3DR directly for inquiry into the issue.

Thanks.

closed 2016-08-07 by ne0fhyk

DarylWM, 2016-08-07:
Sorry, my understanding is [the Services app is in this repo](https://github.com/dronekit/dronekit-android/tree/develop/ServiceApp), so that's why I raised it here. Where should I be raising it?