dronekit/dronekit-android #441: does this dronekit-android work for PX4?

seantobe, 2016-09-12:
Dear all:
I am trying to develop an Android APP that can take off PX4 simulator JMAVSim. I have successfully compiled and run the code using Android Studio( following the link in "http://android.dronekit.io/first_app.html"). Am I doing it right? If I turn on JMAVSim, and run my APP, would it work? Thanks for replying.

BR
Sean Hsu

squilter, 2016-09-12:
The major issue with PX4 compatibility is changing modes. The standard method of changing modes in dronekit-android will not work. You will have to use the `ExperimentalAPI` and send the mavlink packet yourself. You can model it after [this](https://github.com/ArduPilot/pymavlink/blob/984cbd244bc17159a2bbcfb68ec508fab8a9ea6d/mavutil.py#L526)

seantobe, 2016-09-13:
Hi:
3q for replying. I will try the way your recommanded later on. However, I cannot even connect to the JMAVSim by UDP port 14550. Please let me know what the problem is, appreciate it. I posted my code below.
Step 1: make posix_sitl_default jmavsim//start the simulator
Step2: run my APP and press the connect buttun (onBtnConnectTap)//---->cannot connect
==============================================
public void onBtnConnectTap(View view) {
        if(this.drone.isConnected()) {
                this.drone.disconnect();
        } else {
                Bundle extraParams = new Bundle();
                extraParams.putInt(ConnectionType.EXTRA_UDP_SERVER_PORT, 14550); // Set default port to 14550

                ConnectionParameter connectionParams = new ConnectionParameter(ConnectionType.TYPE_UDP, extraParams, null);
                this.drone.connect(connectionParams);
        }
}

IeiuniumLux, 2017-05-02:
I recently tested DroneKit-Android with the PX4 flight stack via the experimental api and it works well as show [here](https://youtu.be/TwOflR7KXuQ).

HansHG, 2017-06-18:
Hello IeiuniumLux,

is it possible to get your test code?

Regards HansHG