dronekit/dronekit-android #442: "Tower application was stopped", same issue on Tower and Tower Beta

zoniiic, 2016-09-12:
Samsung Galaxy S4 GT-I9515
Android 5.0.1, freshly after restoring the system
Tower Beta v4.0.0-beta.4

As stated in the title, problem occurs both on Tower and Tower Beta apps, but since Tower Beta is the new app and under development, I would like to contribute here.

I'm a totally new user, just built my DIY drone based on Pixhawk, never used DP or Tower before. 3DR radios are chineese copies, work very well on PC and MP. After connecting it with OTG to my S4 and pressing "Connect" to Pixhawk, the application shuts down with a message "Tower Beta application was stopped" - something similar, Android OS in Polish language. However it still works in the background (on task bar it's still connected to Pixhawk), but after rejoining the app I have to connect again and the error occurs still. Below I put the crash log.

Not sure if I'm writing on the good forums - if not I'm sorry, I'm totally new to this business. For any help - big thanks in advance! Can't wait to use Tower Beta on my phone as carrying my laptop all the time for telemetry is quite annoying :)

[2016-09-12 13-50-12.txt](https://github.com/dronekit/dronekit-android/files/467240/2016-09-12.13-50-12.txt)

guiseco, 2016-09-22:
Try use google TTS -> https://play.google.com/store/apps/details?id=com.google.android.tts&hl=en

caplan, 2016-09-29:
@zoniiic, could you open a browser on your Samsung, go to http://mybrowserinfo.com/detail.asp?bhcp=1 , and tell us what the `User Language` field is set to?

zoniiic, 2016-10-14:
Sorry guys for lack of replies, I was abroad.
@guiseco TTS is installed and working.
@caplan User Language: Not detectable with JavaScript disabled