dronekit/dronekit-android #447: Tower doesn't ignore other system HB, i.e. sys_id now == 1

billbonney, 2016-09-25:
This is related to #443 since when this is fixed you get the connection bar/mode control toggling between HB from sys_id 1 and other systems, sys_id 254 for example,  that could be connected from a second GCS. 
These images show two frames as the control bar changes (probably has a  different name but I am just getting familiar with the code)
[image]
[image]

billbonney, 2016-09-27:
fixed in version_3.0.2 (but still needs #443 fixed)

closed 2016-09-27 by billbonney

seanmcleod, 2016-09-30:
Won't this fix also filter out broadcast messages now?

billbonney, 2016-09-30:
Which messages are those? I assume you mean sys_id of 0.  It's easy to add a condition for that but if there is no responsder in the drone object for a broadcast messages nothing changes.

seanmcleod, 2016-10-04:
@billbonney Yes I'm referring to broadcast messages which use a sys_id of 0. When we first came across issue #443 I looked into the MavLink spec/docs to understand the valid ranges of sys_ids etc. and noticed the broadcast option.

There may be no responders at the moment for broadcast messages but if someone decides to add code for a broadcast message in future they'll be confused when it doesn't make it through.

billbonney, 2016-10-04:
@seanmcleod I have create an issue (enhancement) to add the broadcast responding feature to MAVLink drone object in Tower.

With the changes above, Tower is closer to supporting multiple drones and ones with a sys_id > 127.

Hope that helps 👍

PR #448 (2016-09-26): Mavlink sysid filter
This fixes #447 where a drone instance picks up messages from other systems that can be on the link
commit 9471662e (2016-09-26): Fixes #447 Add filter on Drone sys_id on incoming messages