dronekit/dronekit-android #450: Solo video decoding glitches

meavydev, 2016-09-29:
I am trying to use dronekit v3.0.1 with the Solo and I am seeing a lot of glitches on the video.
I am seeing traces like:
NaluChunkAssembler: Sequence number is out of order: 49143 != 49148

The 3DR app is fine, so I am wondering if there is some other configuration I am missing?

ne0fhyk, 2016-09-29:
@meavydev starting with v3.0.0 the 3DR app has been deprecated. All the logic is now bundled with the client library. I have seen this issue on some device (nexus 5x) but not others (galaxy s6, nexus 6p), so it might be related to how much processing time the video decoding thread is given.
Bumping up the thread priority might be able to help.

meavydev, 2016-09-30:
OK a rewrite of the threads later and now it's much better, so I think it was taking too long elsewhere. It is actually running on a 6P running Nougat.

closed 2016-09-30 by meavydev

ne0fhyk, 2016-09-30:
Cool! Did you have to bump the thread priority?

meavydev, 2016-09-30:
Yes and it is still not 100%, but much better.
It also degrades after a few minutes on the 6P, as it is getting hot, so could be doing some thermal throttling.
So I don't think it's a dronelink problem, more of a device issue, and tweaking things has improved it.