dronekit/dronekit-android #464: Video Decoding Errror

jasonhamilton, 2016-12-01:
I originally posted this issue on the Tower repo but I believe it has to do with dronekit.  

https://github.com/DroidPlanner/Tower/issues/1836

jasonhamilton, 2016-12-01:
It looks like the Kindle Fire supports its own set of video specifications and sizes.  Perhaps the code should check to see if the video specification is supported on the device prior to starting the stream?

https://developer.amazon.com/public/solutions/devices/fire-tablets/specifications/02-video-specifications

GPSFan, 2016-12-05:
I have had a problem with the Tower/Solo app video decoding on MTK based tablets for a long time. 
The observed behaviour is that Tower can control the Solo properly, but the video is never displayed.
The Solo app may decode 1 frame of video before resetting the connection. Then this repeats.
I hooked up USB debugging and captured the resulting logcat for both the Solo app and Tower. Although the versions of each I used are fairly old (last March for Solo and July for Tower) the problem remains to this day.
Hope the logcat output helps someone. I don't think this is a Tower issue but may be related to dronekit-android or the GLES android drivers for the MTK gpu. (Mali400)
BTW the Solo app did the same thing both before and after 3DR moved away from 
[video_problem.txt](https://github.com/dronekit/dronekit-android/files/630019/video_problem.txt)
[video_problem_tower.txt](https://github.com/dronekit/dronekit-android/files/630020/video_problem_tower.txt)

3DRServices.