dronekit/dronekit-android #482: DroneKit 3.0.+ crashing on attemped connection to tower

epicmittmitt, 2018-04-07:
I'm trying to implement a drone app on Android but am running into this frustrating error. Whenever I try to connect to a tower (that is, call `tower.connect(this)`) my app immediately crashes. When I comment this line, the app runs fine but, of course, won't connect to a drone.

Currently, I'm using Dronekit 3.0.+. Downgrading to 2.9.+ works. Is that my only option? It seems like 3DR is deprecated ...

Here is the error; I can always provide more if needed:

```
android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=null pri=-2 contentView=null vibrate=null sound=null defaults=0x0 flags=0x40 color=0x00000000 vis=PRIVATE)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1784)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6520)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:442)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

chaviw, 2018-04-15:
Notification channels was added in Android O and crashes if not used correctly on apps targeting O. Dronekit has not be updated or tested on O. Try downgrading your target SDK in your Android app to < 26.

sudarshanw, 2021-01-16:
> I'm trying to implement a drone app on Android but am running into this frustrating error. Whenever I try to connect to a tower (that is, call `tower.connect(this)`) my app immediately crashes. When [...]
Notification channels was added in Android O. If you resolve this error then you need to update updateForegroundNotification method inside org.droidplanner.services.android.impl.api.DroidPlannerService file.
[updateForegroundNotification.txt](https://github.com/dronekit/dronekit-android/files/5824319/updateForegroundNotification.txt)