dronekit/dronekit-android #491: Crash NullPointerException SSID

cordlesstuba, 2019-05-09:
Hello !

In Android 9, Tower GCS crash when i'm trying to connect the phone on a Wifi with this error : 

[image]

Could you provide a patch for this error please ?

Thanks !

Saikat2019, 2019-09-07:
what are the compileSdkVersion and targetSdkVersion you are using, First I was using 28, then I got bad notification channel error for startForeground, then I downgraded the compile and targetSdkVersion to 24, then I am getting this error while using android Pie(28)

afarhan39, 2020-01-06:
You need to modify file WifiConnectionHandler.java

Swap this line 
`final WifiInfo wifiInfo = intent.getParcelableExtra(WifiManager.EXTRA_WIFI_INFO);`
to this
`final WifiInfo wifiInfo = wifiMgr.getConnectionInfo();`

HefnySco, 2020-04-28:
check this [https://github.com/HefnySco/dronekit-android/tree/fix_wifi_conn](https://github.com/HefnySco/dronekit-android/tree/fix_wifi_conn)