dronekit/dronekit-android #499: API Reference and Documentation links not working

frank6ix, 2021-01-06:
I tried accessing the API Reference and API documentation links but none worked.
Documentation: http://android.dronekit.io/
API Reference: http://android.dronekit.io/javadoc/

ZanzyTHEbar, 2022-04-17:
Indeed - it's very annoying

hamishwillee, 2022-04-18:
@mrpollo Is this still supported?

ZanzyTHEbar, 2022-04-19:
Yeah, it's still supported. The issue is just the hyperlinks and the website. According to their gitter chat the website has been down for awhile. The repo itself has a folder with a bunch of documentation though - which is the bareminimum to get started. I am using it currently for a project - it does work.

mintheinwin, 2023-07-13:
Documentation links are still not working? Anyone can access?

RossHacquebard, 2023-07-13:
Links are still broken for me also, does anyone have any other source for the API documentation?